dekuNukem/duckyPad #1: Purchase kit or maybe online bill of materials

sammcj, 2017-11-10:
Neat project!

Not a big fan of the LEDs but love the design.

Any chance of selling some kits with everything needed to make one?

I’d not - perhaps one of those fancy online sites where you could enter the bill of materials and people then links to a cart to buy the parts from somewhere?

Thanks!

nicopastorello, 2017-11-10:
I could buy this kit, I found it incredible

RBM-PITT, 2017-11-12:
What is the budget for the project? Where can i get details on the stencil/oven for soldering - I do not have background in that and need some guidance.

dekuNukem, 2017-11-13:
Well currently I'm not planning to produce kits of this project, it's a lot of work and I don't have time at this moment. However I do still have some empty PCBs left over so if any of you want I can sell one to you. I'll also put together a BOM and put it in the repo this week.

RBM-PITT, 2017-11-14:
I'd like a board. how much?

Cabalist, 2017-11-15:
I'd also like a board.  :)

dekuNukem, 2017-11-15:
I dug around the pile of leftovers a bit, and put together this:

https://i.imgur.com/aqwWSpI.jpg

* one empty circuit board
* one translucent diffuser keyboard plate
* one clear acrylic back plate(it looks translucent because of the protective cover, peel it off and it's clear)
* M3 nuts, screws and standoffs 10 each
* 4 anti slip rubber foot

I'm thinking about $20 for the lot, and I have 3 of those bundles available.

It's not an easy undertaking though. You will need to source the other components, and know how to do surface mount soldering to put them together. Technically you can upload the firmware through USB DFU mode without a programmer by holding down the DFU button while plugging it in. Though it's nice to have a st-link programmer just in case, and some experience with Keil STM32 development would be helpful as well.

Anyway, if you still want to go ahead please contact me on dekunukem gmail com and I'll mail them out once payment is received.

Ill finish writing the BOM before the end of this week.

Cabalist, 2017-11-17:
Hey I sent you an email.  Would love to get your kit* if it is still available.  

* Fully aware of the limitations and requirements. :)

closed 2018-02-24 by dekuNukem

sammcj, 2018-02-24:
I take it that by closing the issue - the kit isn’t available?

dekuNukem, 2018-02-26:
I closed this issue as part of my GitHub housekeeping. Currently I don't have a plan for a complete kit, but I do have one last partial set as described in my comment above.