dekuNukem/duckyPad #4: I would like to order PCB's for this project

Avery17, 2018-07-30:
I'm trying to order PCB's so I can build this myself. I went to order through OSH Park but it says someone else owns the project files and has not shared them. Would you be able to share them so I can order them for print?

Thanks

dekuNukem, 2018-07-30:
I haven't used oshpark myself, but to me it looks like you just need to upload the [.brd](https://github.com/dekuNukem/kbord/blob/master/pcb/lul.brd) file on its main page. Anyway, personally I use https://jlcpcb.com/ for all my PCBs, they are faster and much cheaper at the same time. 

Right now you can get 10 PCBs for $2 made in 2 days, a much better deal than oshpark.

Avery17, 2018-07-30:
I tried to upload the .brd file but it seems to have some copy protection built in and won't let me order it since it seems someone else has already. I'll try jlcpcp though! Thanks!

closed 2018-08-01 by dekuNukem