dekuNukem/duckyPad #5: Rotary Encoders does not work.

barbaroskucuk, 2019-09-04:
I could not get rotary encoders to work. I believe its software related. Any help?

dekuNukem, 2019-09-04:
Hi barbaroskucuk:

You're right, I didn't find rotary encoder particularly useful in my usage, so I didn't write code for it in the firmware. They were removed in the new board revision too. 

Still, they are connected to the microcontroller, so if you want you can write something yourself, there are many tutorials on this subject, here's one:

https://howtomechatronics.com/tutorials/arduino/rotary-encoder-works-use-arduino/

Best Regards,

Allen

barbaroskucuk, 2019-09-04:
Thanks so much for the quick response. While you are here i want to ask another question too. Is it possible to change the language in the firmware. What i mean is when my keyboard language is set to english it works perfectly but when i switch to turkish, it does not work properly.

dekuNukem, 2019-09-05:
when you say "not work properly", what does it do that's wrong?

barbaroskucuk, 2019-09-30:
Since i have Turkish layout in my physical keyboard, for example i have " ç " but in English layout its a different letter or a symbol so it does " :" instead of " ç ".  What i'm guessing is that ASCII characters are different for English and Turkish. Is there any way to change the characters or make it language changeable?

LordMayonyies, 2020-08-06:
> I could not get rotary encoders to work. I believe its software related. Any help?

Hii! Did you manage to make it work? I'm looking forward to buy one and I would like to put a pair of rotary encoders. If you manage to make it work, can you share some info please? Thanks in advance!

closed 2020-09-12 by dekuNukem