dekuNukem/duckyPad #6: Build details

disfenoid, 2019-10-04:
Hi, i would like to try to build kbord. As i can see, version described in readme is different from current version of the board. So i have a few questions:
* Is current version of the board is working, or i should use version from specific commit?
* I noticed that now usb type-c used instead of micro, is there any other changes in BOM?
* I'm not a big fan of rgb, will the board work if i will just not solder them?

dekuNukem, 2019-10-04:
Hi disfenoid:

The new version of the board is fully working, I have been using one myself. 

As for the BOM, USB connector is different, the four pushbuttons are different, and rotary encoders are removed. I think everything else is the same. I'll take a photo of the new board and update the BOM later today. 

The board will work fine if you don't solder the RGB LEDs

dekuNukem, 2019-10-04:
Here are the photos of the current version:

[image]

[image]

I'll update the BOM tomorrow.

disfenoid, 2019-10-05:
Great, thank you!

dekuNukem, 2019-10-05:
BOM is here: https://github.com/dekuNukem/kbord/blob/master/pcb/kbord_v2_bom.xlsx

NeverMinde, 2019-11-13:
> BOM is here: https://github.com/dekuNukem/kbord/blob/master/pcb/kbord_v2_bom.xlsx

Hi, I am trying to build kbord but I got some questions about oled screen connector in 2nd version of your PCB. The 30pin wire bus of 0.96" oled screen have  got 1mm step but PCB pads have got 0.5mm step. So finally question:  
How do you solve this problem?
Can you attach some photo of backside with connected screen?

disfenoid, 2019-11-21:
@Ep1quell, I ordered this display and connector: aliexpress.com/item/4000375678671.html, but didn't try yet (still waiting for delivery).

Ep1quell, 2019-11-22:
@disfenoid , with your help I found one more link, here it is: https://s.click.aliexpress.com/e/bd6hKgeO
The main difference is driver IC. In first version of BOM deku said that he using oled with ssd1306 so I think that this will be better cause there is SH1106 in your link

dekuNukem, 2019-11-22:
This one looks like the OLED that I used:

https://www.aliexpress.com/item/32950730252.html

1.3 inch, 30pin FPC connecter, SSH1106 controller.

dekuNukem, 2019-11-22:
this listing comes with the FPC connector as well, and also detailed drawings:

https://www.aliexpress.com/item/4000350633981.html

make sure the 30-pin FPC connector is top-contact type,

kow, 2020-01-10:
do you have any details on the new buttons and usb connector?

dekuNukem, 2020-01-14:
you can find it here:

https://github.com/dekuNukem/kbord/blob/master/build_it_yourself.md

I significantly updated build details today, hope that helps!

closed 2020-01-14 by dekuNukem