dekuNukem/duckyPad #8: switch plate thickness

irongu, 2019-12-10:
Hi, I'm trying to build a kbord, how thick material should I use when making the switch plate?

And I noticed that you changed the model of oled in version 2, but the new model was not described in the bom, which made me buy the wrong oled. After I got the parts, I found there was a problem, and I didn't know the information of the new oled until I came back to see issues. So I hope to add a description of the new oled in the bom so that the people behind will not make mistakes again. Thank you.

ftrrarr, 2020-01-14:
MX plates are usually 1.5 mm

dekuNukem, 2020-01-14:
I used a 2mm acrylic plate and it's fine. I guess 1.5mm works too.

closed 2020-01-14 by dekuNukem

dekuNukem, 2020-01-14:
I have also updated the documentation significantly today with much more build details, hope that helps!