dekuNukem/duckyPad #10: kbord does not work if 3.3V pin not connected to STLink

ftorresarreola, 2020-01-14:
I had a board made, flashed it using the instructions and I get nothing from connecting it to USB C, but if I connect the 3.3V Stlink pin, it works fine. What could be going on here?

dekuNukem, 2020-01-14:
I would make sure that the pins on the USB connecter are properly soldered and not touching each other. Also make sure all the pins on the microcontroller is properly soldered and has contact with the pad, sometimes they look like they are soldered but isn't actually touching. 

And if there is nothing when plugged into USBC but works when 3.3V is provided, there might be something going on with the power regulator, so check that it is getting power and outputting the proper voltage, also check the decoupling capacitors are not soldered backwards.

ftrrarr, 2020-01-15:
What are the designators for the decoupling capacitors? That's probably the culprit since I thought SMD capacitors were unpolarized.

dekuNukem, 2020-01-15:
The smaller 0805 ceramic capacitor are unpolarized, however the big tantalum capacitor is. 

It is C27 near the top right corner:

https://raw.githubusercontent.com/dekuNukem/kbord/master/resources/pics/pcb_front.jpg

The side with a brown band is the positive side.

ftrrarr, 2020-01-15:
Can you link to one like that? I am having trouble finding one with the specs on the BOM. I am so close.

dekuNukem, 2020-01-15:
Here's one:

https://www.ebay.com/itm/20pcs-1210-25V-22uF-226E-B-type-3528-SMD-tantalum-capacitors/292219373942

You can search "1210 tantalum capacitors" on ebay and there are a bunch of them.

ftrrarr, 2020-01-15:
I think there's something wrong on the BOM? It says 47 uF 1220 tantalum cap but I think it should be 22uF 1210 tantalum cap?

dekuNukem, 2020-01-15:
The exact value of C27 doesn't matter that much, anything above 10uF would do. Its just to stabilize the input to the regulator, you might even be able to leave it off, but it's not recommended.

The 1220 is indeed a typo though, should be 1210.

ftrrarr, 2020-01-15:
I've removed the cap but I still have the same problem where I can't get it to work without connecting the 3.3V pin. Would adding the polarized one fix it or is there other troubleshooting I can do?

dekuNukem, 2020-01-15:
Take a look at the pinout of the 1117 voltage regulator (page 3):

http://www.ti.com/lit/ds/symlink/lm1117.pdf

if you have a multimeter, measure the input and output voltage of the regulator. The input should be at USB voltage of 5V, and output should be 3.3V.

ftrrarr, 2020-03-05:
I finally got to measure the regulator, the input gives me 4.77 V which should be fine? But the output is reading at 1.26 V.  One with C27 removed is 1.26V but one with an unpolarized cap on C27 gives me 1.5 V as the output. Is this due to the lack of the tantalum cap? Or what could be the issue? I used REG1117 for the voltage regulator

dekuNukem, 2020-03-05:
Are you sure the 1117 regulator is a 3.3V variant? They have all kinds of different output voltages and 1.2V is one of them.

ftrrarr, 2020-03-11:
That was it. It works now. This was fun! thanks for the project and the help, it's very much appreciated.

closed 2020-03-11 by ftrrarr