dekuNukem/duckyPad #12: Internationalization

simonCor, 2020-05-11:
duckyPad works fine - as long as you have an english keyboard layout. The problem is briefly discussed here: https://github.com/dekuNukem/duckyPad/issues/5

The problem is, that the OS assumes a keyboard layout (and that the USB HID spec not just allows UTF-8). Therefore when e.g. a german keyboard layout is set, the scancodes are interpreted differently.

The keyboard firmware QMK has already a solution in place for this, see https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_extras/

It is a bit clumsy because then it is necessary to configure the layout used in the firmware but IMHO it is the most practicable solution.
I suggest to modify keyboard.c/.h and to port the QMK solution to duckyPad.

Any comments on that? Others already working on that?

simonCor, 2020-05-20:
This shows the first attempt on my blackpill port:
https://github.com/simonCor/poor-mans-ducky-pad/commit/cd960a5336f6534f1f1e5d817e6d8283d3e78115

Should be useable 1:1 on the original.

This only fixes the ascii characters in different languages - adding more non-ascii characters (äöüß) is not as trivial - then we also have to discuss input format (e.g. UTF-8) and how far we want to go by translating this to keycodes...

dekuNukem, 2020-10-22:
I have added support for AZERTY and Dvorak keyboard layout in a recent update, thanks for the tip!

closed 2020-10-22 by dekuNukem

dekuNukem, 2021-08-19:
Let me do some testing and report back. Do you have a sample script that you would like me to test with?

M-14K, 2021-08-20:
of course, i will let you my profile test examples

[example.txt](https://github.com/dekuNukem/duckyPad/files/7021878/example.txt)

[dp_settings.txt](https://github.com/dekuNukem/duckyPad/files/7021884/dp_settings.txt)
[dp_stats.txt](https://github.com/dekuNukem/duckyPad/files/7021885/dp_stats.txt)
[README.md](https://github.com/dekuNukem/duckyPad/files/7021886/README.md)
[key2.txt](https://github.com/dekuNukem/duckyPad/files/7021887/key2.txt)
[config.txt](https://github.com/dekuNukem/duckyPad/files/7021888/config.txt)

dekuNukem, 2021-08-21:
It seems to work fine on mine, here's a short video:

https://www.youtube.com/watch?v=0jcJJhw7vBk

I would suggest make sure Windows 10 is set to French keyboard layout, and also select French layout on duckyPad, make sure it is on [latest firmware](https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md) too.

M-14K, 2021-08-21:
I tryied to downgrade and then upgrade to the last firmware, changed the software too.. 

My set is on french layout only but i tryed to add us and sadly the error stil persist

[image]
[image]

i took a shot to show you how it works in both of my 3 operating systems... ( all on french only)
https://youtu.be/qBAS8buCyC8

dekuNukem, 2021-08-22:
I seem to remember another user with similar issue solved by using a different USB port for some reason.

It would be helpful if you can test it out on a different computer as well, just to rule it out.

dekuNukem, 2021-08-22:
You can also try clear the keymaps:

Mount the SD card on the PC, delete the `keymaps` folder.

With the SD card still mounted, launch config app, load the SD card, select keymap again, and save. See if it works.

dekuNukem, 2021-08-22:
you can also give this beta firmware a try:

https://github.com/dekuNukem/duckyPad/raw/master/firmware/beta_0.19.6.dfu

I adjusted how keystrokes are sent, so it might be a bit more compatible.

M-14K, 2021-08-22:
"You can also try clear the keymaps:
Mount the SD card on the PC, delete the keymaps folder.
With the SD card still mounted, launch config app, load the SD card, select keymap again, and save. See if it works."

It works but there is a catch, i dont know why but after doing this step i just mounted the SD card again
I created a new profile test and wrote " STRING Câble" to test the "â"  letter.
It worked, but when i just put the old profiles on the SD card the problem apears again i dont lnow why..
Some of them just apeard blank

I guess that i will need to build the profiles 1 by 1 again to bypass this issue

Anyway thanks for your precious time, at least i know that i can make it work again :)