dekuNukem/duckyPad #21: Keys using HOLD getting stuck on

madwort, 2020-11-10:
Having some issues with the `HOLD` functionality - if I hold for a brief period it works great, if I hold for a longer time it appears to key wedged on. It doesn't seem to be a mechanical issue - unplugging & replugging the duckypad resets it and the key is no longer held on. Using firmware `0.8.0`.

dekuNukem, 2020-11-10:
Thanks for letting me know. I'll look into this and do more testing.

dekuNukem, 2020-11-10:
A couple of quick questions 😅 

Does the `HOLD` command issue happens with every key?

How long do you have to hold it for the issue to appear?

Which operating system are you using?

Could you provide a sample script?

madwort, 2020-11-10:
The `HOLD` issue happens with at least two keys, I haven't tested any others yet.

I don't have to hold very long for it to appear. 

This was whilst using macOS. However, the button in question is yellow normally and goes blue when I press it (then fades back to yellow). When it gets stuck the light is also stuck on blue, which makes me think the duckypad is stuck not the OS. 

This issue was observed with these two scripts:

`HOLD MK_VOLDOWN`
`HOLD MK_VOLUP`

madwort, 2020-11-10:
Hmm, weird - it reproduced consistently earlier, but now I can't get it to reproduce!

madwort, 2020-11-10:
Weird, I could trigger this issue consistently earlier, but I'm trying again now & I can't make it do it.

dekuNukem, 2020-11-10:
Actually I was able to reproduce with volume up/down. Those media keys have different hold and release mechanisms, and maybe I missed something. I'll look into it.

In the meantime, don't use `HOLD` with media keys.

You can invoke them regularly and hold the key down. It will still repeat.

dekuNukem, 2020-11-10:
Check discord for new firmware.

madwort, 2020-11-11:
Loaded `0.10.0`, now to see if I can make it break...

madwort, 2020-11-11:
I've not managed to make it stick, hopefully it's sorted! Thanks!

closed 2020-11-11 by madwort

dekuNukem, 2020-11-11:
That's great to hear! If you have time try out the other profiles and keys as well, hopefully I didn't break something else.