dekuNukem/duckyPad #23: SW color compatibility PC

PiTronica, 2020-12-20:
Hi,

Is it possible to add coloroption to the PC configurator for the up and down key.
Python programming is a bit out of my leage, i got the code working in the STM, all i need is an output in the config file like:

SWCOLOR_16 255 255 255
SWCOLOR_17 255 255 255

i tried myself but failed miserably and should have taken a copy of my sd card first -_-

PiTronica, 2020-12-28:
I got it to work as far as it takes the background color.
But it would be handy to be able to set the color from the sd card so if its possible to add this in the pc software?

jose1711, 2023-03-11:
Not sure I follow. There are 15 LEDs on DuckyPad and yet you are trying to configure extra 2?