dekuNukem/duckyPad #24: HOLD multiple keys at the same time

de-luca, 2021-01-12:
Would it be possible to `HOLD` multiple keys at the same time?

My usecase is to use the pad as an OBS deck.
However when OBS is out of focus the input delay miss fire hotkeys and make it unreliable.

Inputing the hotkey by hand on the keyboard works so a more "natural/human" holding could work.

By using `HOLD` not the intended way it works once then will run the last `HOLD`.
```
HOLD CTRL
HOLD SHIFT
HOLD ALT
HOLD BACKSPACE
```

dekuNukem, 2021-02-24:
`HOLD` command is originally intended for scenarios like WASD gamepad and push-to-mute. Normally when a key is pressed, the microcontroller reads its file in SD card and execute it line by line. However, this isn’t fast enough for real-time WASD gaming in my testing, so `HOLD` command actually caches the key to memory after the first press, thats why it only works with one key for now.

For multiple keys, you can probably use `DEFAULTDELAY 100` to add more time between each line of script, and  `DEFAULTCHARDELAY 100` to add more delay between each keystroke. In your example it would be something like:

```
DEFAULTDELAY 100
DEFAULTCHARDELAY 10
CTRL SHIFT ALT BACKSPACE
```

I haven’t tried this myself, so feel free to let me know how it goes. 

I might also add an uncached HOLD command in an upcoming update, when things slow down a little.

de-luca, 2021-02-27:
Thanks a lot @dekuNukem! `DEFAULTCHARDELAY` was the solution for my usecase.
```
DEFAULTCHARDELAY 50
CTRL SHIFT ALT BACKSPACE
```

I would have not found out about `DEFAULTCHARDELAY` effect on single line modifiers by myself.

I tested with your solution but in my case it had some strange behaviour on windows and kept the CTRL signal up.
By trial and errors, the DEFAULTCHARDELAY I found a timing that was triggered in OBS.

closed 2021-02-27 by de-luca