dekuNukem/duckyPad #25: LOOP not running ENTER after all cases exhausted

de-luca, 2021-01-12:
Here is my test loop:
```
LOOP0:
STRING first action
STRING FOO
DELAY 500
ENTER
SWCOLOR 128 0 0

LOOP1:
STRING second action
STRING BAR
DELAY 500
ENTER
SWCOLOR 0 128 0
```

The first LOOP0 and LOOP1 run correctly and swap color.
After the first "loop" no ENTER are inputed.
The SWCOLOR is runned however.

output:
```
first actionFOO
second actionBAR
first actionFOOsecond actionBARfirst actionFOOsecond actionBAR
```

The script was saved from configurator 0.9.0 on windows and tested on Windows 10 and MacOS 11.1.

dekuNukem, 2021-01-13:
Thanks for letting me know, I'll try it out later today.

dekuNukem, 2021-01-13:
I can't seem to reproduce on both Win10 and macOS Big Sur.

I am on firmware 0.13.0, PC app 0.9.0.

https://www.youtube.com/watch?v=rVSTN8BxWlM

If your firmware up to date? You can check by removing the SD card (as shown in video). If not, try [updating it](https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md)

Another possibility is the duckyPad is typing too fast for your computer, you can slow it down with various `DELAY` commands, which you can [find here](https://github.com/dekuNukem/duckyPad/blob/master/duckyscript_info.md#defaultdelay).

dekuNukem, 2021-01-13:
This is strange, does the `ENTER` key issue happens with other scripts, or only with `LOOP` command?

de-luca, 2021-01-13:
I made a script with a bunch of ENTER and STRING like:
```
STRING 1
ENTER
STRING 2
...
```

The ENTERs are okay in that case.

dekuNukem, 2021-01-13:
what if you put two `ENTER`s in the loop script? Maybe try it on a few different computers? It is really strange indeed.

de-luca, 2021-01-15:
After testing with all the computers in the house and all the different usb ports, I finally found the problem.
It was usage.

I did all my tests in a text editor which would try to complete input on the fly and eat some enter and tabs input.
So yeah, PEBCAK...

Thanks for the help! And sorry for the anti-climatic end.

closed 2021-01-15 by de-luca