dekuNukem/duckyPad #30: User contributed scripts

jadjoubran, 2021-02-03:
Hey,
I just received my DuckyPad, it's awesome, thanks!

Is there a list of user contributed scripts? It would be nice to have somewhere where a list can be compiled so that we can get inspiration

dekuNukem, 2021-02-03:
Hi there! An user has made a website for script sharing, it is still at its early stages though, feel free to check it out: https://duckypad.io 

You can also take a look at the sample profiles that came with the SD card, which covers a lot of the usages.

Finally, there is also the [discord chatroom](https://discord.gg/4sJCBx5), and people will be happy to help you there. :)

jadjoubran, 2021-02-03:
Yes that's exactly what I've been looking for, awesome! Maybe worth adding to the readme?

jose1711, 2023-03-11:
The domain seems to have expired :disappointed: