dekuNukem/duckyPad #31: configurator unusable on high res displays?

blake5634, 2021-02-06:
configurator looks like this 
[image]
Seems like lot of hard coded pixel values in the code.   Adding size to these dimensions:  
```
THIS_VERSION_NUMBER = '0.9.0'
MAIN_WINDOW_WIDTH = 800    # increase to 1500 eg
MAIN_WINDOW_HEIGHT = 600
MAIN_COLOUM_HEIGHT = 533
```
doesn't help.

dekuNukem, 2021-02-06:
It looks like some kind of linux system, can you let me know what distro are you using?

blake5634, 2021-02-06:
kubuntu 18.04 / Dell XPS-15

Thanks for your help!
Blake

blake5634, 2021-02-08:
One more update.   I set my laptop display to 1920x1080 and it didn't fix the problem.

dekuNukem, 2021-02-08:
Thanks for letting me know! Now that I have finished shipping, I'll finally have time to investigate some bugs.

I'll install kubuntu on my laptop and try it out.

dekuNukem, 2021-02-10:
I tried kubuntu 18.04 live USB on my Dell G5, it seems to work fine on its 1080p screen:

[image]

From clean install, the only thing I did was install tkinter with `sudo apt install python3-tk`, then I launched the app with `python3 duckypad_config.py`. I don't have a 4K screen to test with currently.

dekuNukem, 2021-02-10:
I think I figured it out, it was the scaling setting in display settings. It works fine when set to 1, but any higher the text becomes too big. I'll look into it, in the meantime you can set the scaling to 1 to use the app.

[image]

blake5634, 2021-02-10:
Brilliant.  Thanks for tracking that down.  Indeed, I had to adjust scale because with so many pixels, default fonts are just so tiny!   BTW, I'm working on a workaround which could help others with configurator issues.  Stay tuned

blake5634, 2021-02-19:
Workaround [here](https://github.com/blake5634/Ducky-Sheet).   It's a spreadsheet template to enter the keypad info, plus a python script that outputs a profile folder with all the needed stuff.

blake5634, 2021-02-21:
confirming:  changing scale back to 1:1 works(!)   (I didn't realize that KDE allows separate scale setting for each monitor)  Thanks!

closed 2021-02-21 by blake5634