dekuNukem/duckyPad #32: Doesnt turn on

Chieis, 2021-02-09:
I was leaving my macropad to chill and then it suddenly turned off. I tried plugging the USB-C connector to different slots and they all produced the same results. TLDR; it wont turn on.

dekuNukem, 2021-02-10:
Hi Chieis:

Try using a different cable, also try power it on without the SD card and see if the screen lights up.

If it was indeed the hardware I can send you another one.

Allen

Chieis, 2021-02-20:
I have tried to do these but it still doesnt turn on. 

Ive used the cable provided in my backing and another one that my friend had owned as well. It does not turn on. 

Can you provide your email so I can send you my address?

Thank you.

dekuNukem, 2021-02-21:
sure! it's dekunukem gmail com

closed 2021-04-04 by dekuNukem