dekuNukem/duckyPad #34: DuckyPad configuration tool -> Windows button

svrooij, 2021-02-15:
The button in the config tool called `WIN` does add `WIN` to the key config.
It then says their is something wrong with the code, and in the docs is said that I need to type `WINDOWS` to send the windows key.

So I thing the button in the app can be called the same but should output `WINDOWS`.

dekuNukem, 2021-02-16:
yeah I was trying to clean up the code a bit and didn't notice that. 😓

Thanks for letting me know, I'll fix it in the next update.

svrooij, 2021-02-16:
@dekuNukem no problemo, I really like the duckypad!

Just a thought, what if the buttons and the allowed codes (like `DP_SLEEP` and `GOTO_PROFILE`) where dynamic? As in, at the first start the app loads a json file (from Github Pages) and then configures the buttons accordingly. As well as the code checking in the app.

Then you can even make it periodically check for a new version. That would stop the need to every time update the config app.

dekuNukem, 2026-02-26:
It should be possible with the latest update actually!

closed 2026-02-26 by dekuNukem