dekuNukem/duckyPad #35: LOOP: Lost of counter increment after waking up from sleep mode

TheBigFudge, 2021-02-19:
Hey @dekuNukem 

I received my DuckyPad from the Kickstarter campaigna few days ago and I am really excited. I only noticed one small problem:

Currently I using the experimental LOOP function to track if the button was pressed the first time or the second time. To trigger different actions each time.
Unfortunately I noticed that after the DuckyPad was in sleep mode, the counter for the loop starts again at 0 and does not keep the last value.

Is it possible to reload or keeping the old counter value for the loop after waking up?

dekuNukem, 2021-02-19:
Nicely spotted! It does reset everything once it goes to sleep, I'll look into this issue in the next firmware update.

TheBigFudge, 2021-02-22:
Great to hear that :)
the same thing seems to happen when you switch between profiles. But I don't know if it is possible here to save the counter variable persistently, when you are switching between modes

dekuNukem, 2021-04-04:
fixed in the latest firmware update, feel free to try it out!

closed 2021-04-04 by dekuNukem

TheBigFudge, 2021-04-06:
seems to work very well. Thanks a lot :)