dekuNukem/duckyPad #36: DuckyConfigurator saving speed

jfbethlehem, 2021-02-22:
When saving the data for the duckypad configuration on SD card, the DuckyConfigurator program hangs for quite a long time, up to half a minute. After this very long duration of saving configuration files, it returns back to normal. This happens even when the amount of 'pages' in the duckypad is very low (4-6).

Expected behaviour: Still functioning application during saving / Quicker saving of files

dekuNukem, 2021-02-22:
When you press `Save` button, it writes data to the SD card. So if hangs for a long time it means the SD data write is very slow for some reason.

If possible feel free to try a different SD card reader, different USB port, or maybe a [different SD card](https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card) altogether and see it helps.

jfbethlehem, 2021-03-08:
Optional change: Maybe only save the files/pages that have been edited. This way, less files will be written to disk & saving will be done faster.

I tried saving to a different disk, but even then it can sometimes hang for quite a long time (10-30 secs) considering the size of the files being written (max 5k total file size with 10 full pages of configurations, mostly keypresses, some longer scripts).