dekuNukem/duckyPad #37: Feature request: Editting the config while the SD card is in the duckypad

jfbethlehem, 2021-02-22:
I would like to edit the configuration of the duckypad while the duckypad is connected to the computer. 

This would save a lot of time: removing the card, putting the card in the reader, putting the reader in the laptop/computer, opening the configurator or config files, editting & saving, unmounting, removing the reader, removing the card from the reader, putting the card back in the duckypad and connecting the duckypad. 

If all the physical work could be skipped by editting directly on the duckypad, it would be a lot easier. it might be possible to do this by putting the duckypad instead of on keyboard mode into disk mode. the original hak4 rubber ducky had a firmware update that allowed this. Maybe something similar could be done with the duckypad.

dekuNukem, 2021-02-23:
That's indeed a good idea, and I thought about it from time to time too. The issue is that it would take a lot of coding, and there just isn't much space left on the microcontroller currently. Its RAM is over 90% full. 

I might try a proof of concept when things slow down a bit, but don't get your hopes up for now 🙂

dekuNukem, 2021-06-01:
I added this feature in the latest beta test, feel free to try it out here:

https://gist.github.com/dekuNukem/3d4ca966c83dca38ae89ee1142806ef2

closed 2021-06-01 by dekuNukem