dekuNukem/duckyPad #40: Firmware Makefile?

gregkh, 2021-03-09:
I was wanting to make some firmware changes to clean up a few tiny things I noticed, but I don't
seem to be able to find a Makefile at the root of the firmware tree that would allow me to build
the code again.

Is it just not checked in?  Any hints on how to rebuild the code would be appreciated.

dekuNukem, 2021-03-12:
Discussed over email. Hopefully that solves the issue!

Let me know if you have further questions.

closed 2021-03-12 by dekuNukem

dekuNukem, 2022-02-16:
Not sure why I went through email for this, anyway, I used Keil IDE for this project, which handles all the builds and makefiles. It is free for the STM32 chip that this uses, and I wrote a [tutorial on how to get started](https://github.com/dekuNukem/STM32_tutorials), first lesson is enough to get it set up.

13rac1, 2022-02-17:
Ah ha! Thanks!