dekuNukem/duckyPad #41: MENU sends wrong key codes

harmathy, 2021-03-10:
I assigned `MENU` to a key, but the expected context menu didn't work. I checked on Linux the key codes using `showkey --scancodes`

* On press: `0x1d 0x38 0x36 0xe0 0x38`
* On release: `0x9d 0xb8 0xb6 0xe0 0xb8`

I checked with another keyboard, and the menu key there results in the following key codes:

* On press: `0xe0 0x5d`
* On release: `0xe0 0xdd`

I also checked with `libinput debug-events`:

The menu key don the duckypad gave:

```
-event5   KEYBOARD_KEY     +22.833s	*** (-1) pressed
 event5   KEYBOARD_KEY     +22.833s	*** (-1) pressed
 event5   KEYBOARD_KEY     +22.833s	*** (-1) pressed
 event5   KEYBOARD_KEY     +22.833s	KEY_RIGHTALT (100) pressed
 event5   KEYBOARD_KEY     +22.977s	*** (-1) released
 event5   KEYBOARD_KEY     +22.977s	*** (-1) released
 event5   KEYBOARD_KEY     +22.977s	*** (-1) released
 event5   KEYBOARD_KEY     +22.977s	KEY_RIGHTALT (100) released
```

the real keyboard:

```
-event10  KEYBOARD_KEY     +27.588s	KEY_COMPOSE (127) pressed
 event10  KEYBOARD_KEY     +27.775s	KEY_COMPOSE (127) released
```

dekuNukem, 2021-03-12:
Thanks for letting me know, I'll do some testing and report back.

dekuNukem, 2021-04-04:
Fixed in the latest firmware update, feel free to try it out!

closed 2021-04-04 by dekuNukem

harmathy, 2021-04-09:
Works like a charm! Thank you!