dekuNukem/duckyPad #42: Switching profile based on focused application

harryjubb, 2021-03-11:
Hi, really cool project!

Wanted to ask if it is possible to switch the selected profile based on the active application?

A use case example:

> As a programmer, I want duckyPad to be able to switch to a specific profile for VSCode snippets when I switch to VSCode, so that I can instantly have access to the macros that I need to insert snippets

dekuNukem, 2021-03-12:
Not at the moment, but I do want to look into it in the near future.

It would be a bit tricky programming-wise, and there isn't much space left on the microcontroller, but I'll see what I can do.

dekuNukem, 2021-06-01:
Added! try it out here:

https://github.com/dekuNukem/duckyPad-profile-autoswitcher

closed 2021-06-01 by dekuNukem

harryjubb, 2021-06-01:
Amazing! Will do: thank you!