dekuNukem/duckyPad #49: WIN10 home "Unknown USB Device"

Olfried2, 2021-03-30:
Hallo, 
after the update yesterday, windows doesn't like the duckypad anymore. In device manager the pad will be (not) recocgnized
as "Unknown USB Device (Device descriptor request failed)"
OS-Info
Version 2004
OS build 19041.867
Experience Windows feature experience pack 120.2212.551.0

Duckypad FW 0.16.0

Any ideas how to get it back to work ?
Thanks in advance

dekuNukem, 2021-03-30:
Hi there! It looks more like a cable issue than the pad itself. Here's a few things to try out:

* Try a different USB cable

* Make sure it's plugged in all the way

* Plug it directly into the computer instead of through a USB hub

* Try it on a different computer

Let me know how it goes!

dekuNukem, 2021-03-30:
By the way, it is not possible to brick the device, so you can always go through the procedure flash any version of the firmware you like.

Olfried2, 2021-03-30:
It seems that a different cable will do the job, I'll investigate further behaviour.