dekuNukem/duckyPad #51: Single key port doesn't work (not bent pins)

dayne, 2021-05-05:
I have a single key that doesn't work on the DuckyPad I recently received.

I double/triple checked for bent pins and swapped keys even.  Decided to by-pass the whole "switch" and jump the contacts with a wire.  I verified the working keys work with the contact jump method but the single key does not work that way.  

Any insight on where to begin the next phase of debugging would be apricated.

dekuNukem, 2021-05-05:
Hi there! Have you tried jumping the metal parts of the hot-swap socket from the backside like this?

https://i.imgur.com/GkvqPQW.png

If it works, the socket should be fine. If not, the check if the socket has lifted from its pad. let me know how it goes.

dayne, 2021-05-18:
@dekuNukem Thanks for the response. Sorry for lag on my side.

Finally dug out the keypad and tried this test at those specific metal parts of the hotswap socket section. Sadly still didn't work.  

Recommendations for next steps?  I was thinking getting out the soldering iron but I want to make sure I'm point in a healthy direction before messing up something that can be easily fixed.

dekuNukem, 2021-05-18:
hey no problem! If that's the case I can send you another circuit board, feel free to DM me on [duckyPad discord](https://discord.gg/4sJCBx5), or email dekuNukem^gmail com, and we can discuss the details.

closed 2026-02-27 by dekuNukem