dekuNukem/duckyPad #53: Duckypad Configuration Tool thoroughly hangs windows explorer session it is launched from

DrMiaow, 2021-05-20:
I launch it and nothing happens. The windows explorer session it is launched in is totally locked up until I reboot my PC. I am using windows 10.

[image]

Any ideas please?

M-14K, 2021-05-20:
I had a similar issue on version 0.12.0 of duckypad.
have you downloaded the newest one 0.12.1?

DrMiaow, 2021-05-20:
Yes. I have resolved it though. I right clicked on it and brought up properties and realised that Windows had blocked it (unsigned and downloaded from internet). I unlocked it and all good now. 

[image]

closed 2021-06-01 by dekuNukem