dekuNukem/duckyPad #58: HID Read Error for USB connection with Configurator

snchao, 2021-06-05:
Can't seem to get this function working in the Configurator and I get the error message below. I can still connect my SD card to my tower without issue, but was curious to see if there's a way to get around this.

Of note, I have also tried connecting the duckyPad through a hub and directly into my tower.

---------------------------
Error
---------------------------
error:

Traceback (most recent call last):
  File "duckypad_config.py", line 1268, in t1_worker
  File "hid_op.py", line 132, in dump_from_hid
  File "hid_op.py", line 90, in duckypad_list_files
OSError: HID read error or busy

---------------------------
OK   
---------------------------

dekuNukem, 2021-06-05:
Make sure your duckyPad is at latest beta firmware:

https://github.com/dekuNukem/duckyPad/raw/master/firmware/beta_v0.19.1.dfu

Follow this guide to flash it:

https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md

snchao, 2021-06-06:
My mistake, I did miss the beta firmware. 

Thank you very much for the help!

closed 2021-06-06 by snchao