dekuNukem/duckyPad #60: Better build for linux

Stiralbios, 2021-06-19:
The build for linux is quite bad. It's only the python files. I tried to use it and I have some issues. Either with with the versions of the module or the version of the lib on the system.
A requirement.txt and a README maybe even a start.sh would be usefull.

joelnb, 2021-06-23:
I agree with this. I have used the Linux version fine before but I recently wanted to use it again and am met with this:

```
AttributeError: module 'hid' has no attribute 'device'
```

I believe this is because the version of the `hid` module I have installed (latest from [here](https://pypi.org/project/hid/) but I also tried the older version) is not what is expected. Is this even the right module? It is not mentioned in the `setup.py` or anywhere else so hard to tell.

I have not been able to find any guide for running this on Linux - I got it working before by reading the code + some trial & error.

joelnb, 2021-06-23:
I put some changes in https://github.com/dekuNukem/duckyPad/pull/61 which allowed me to get things running on Linux. The UDEV rule is required for me - it never connects without that.

This allows the connection to complete and I can see the configuration the device is using but I have not been able to save any changes - I don't see an error & the files do get created but the content is never correct.

joelnb, 2021-07-05:
Hey @Stiralbios, could you please give it another go with the latest version from the git repo after having a read of this https://github.com/dekuNukem/duckyPad/blob/master/app_posix.md

I am able to run the app on Linux using this method so hopefully it works for you too!

Stiralbios, 2021-07-19:
@joelnb 
Thanks, the build work.
However for some reason the udev rule for the profile editing seems to be not enought for me. I checked with lsusb and I have the right idproduct and idvendor. I didn't had time to investigate it more

hasufell, 2022-01-13:
> Hey @Stiralbios, could you please give it another go with the latest version from the git repo after having a read of this https://github.com/dekuNukem/duckyPad/blob/master/app_posix.md
> 
> I am able to run the app on Linux using this method so hopefully it works for you too!

Doesn't work here.

The only way to get the app running is this:

```bash
cd pc_software
python -m venv venv
. venv/bin/activate
pip install appdirs hidapi
python duckypad_config.py
```

which is not in the linked file `app_posix.md` file anymore.

Trying to connect to the device results in:
[image]

----

Also tried as root. Same story.

joelnb, 2022-01-25:
@hasufell I think in that case it is actually connected to the device (so the udev rule is probably working on your system) but something else is wrong. If I were you I would try looking at the output with `HID_OP_LOG=debug` - that should tell you some more about the situation but it can be fairly hard to understand.

hasufell, 2022-01-27:
> @hasufell I think in that case it is actually connected to the device (so if I'm right the udev rule is working on your system) but something else is wrong. If I were you I would try looking at the output with `HID_OP_LOG=debug` - that should tell you some more about the situation but it can be fairly hard to understand.
> 
> FWIW I can still connect using the latest code & the commands you used.

That doesn't tell me much:

```
2022-01-27 19:27:09,334 [DEBUG] hid_op: Found duckyPad: {'path': b'1-1.4:1.0', 'vendor_id': 1155, 'product_id': 53532, 'serial_number': 'DP20_78256f7f', 'release_number': 512, 'manufacturer_string': 'dekuNukem', 'product_string': 'duckyPad(2020)', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
```

It complains via GUI sometimes that it needs an udev rule. But that rule is already present as per: https://github.com/dekuNukem/duckyPad/blob/master/app_posix.md#udev-rule

hasufell, 2022-01-28:
I also have to note that [duckyPad profile autoswitcher](https://github.com/dekuNukem/duckyPad-profile-autoswitcher) is able to connect and run correctly.

thismatters, 2022-03-09:
I'm at the same error as @hasufell. I've added the udev rules and installed the other dependencies. I'm getting this in the debug:

```
2022-03-09 13:30:54,935 [DEBUG] hid_op: Found duckyPad: {'path': b'1-1.4:1.0', 'vendor_id': 1155, 'product_id': 53532, 'serial_number': 'DP20_782f6000', 'release_number': 512, 'manufacturer_string': 'dekuNukem', 'product_string': 'duckyPad(2020)', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
2022-03-09 13:30:54,944 [DEBUG] hid_op: Found duckyPad: {'path': b'1-1.4:1.0', 'vendor_id': 1155, 'product_id': 53532, 'serial_number': 'DP20_782f6000', 'release_number': 512, 'manufacturer_string': 'dekuNukem', 'product_string': 'duckyPad(2020)', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
2022-03-09 13:30:55,078 [DEBUG] hid_op: duckypad_list_files: fname=('SYSTEM~1', 1) result=[4, 0, 0, 1, 83, 89, 83, 84, 69, 77, 126, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,095 [DEBUG] hid_op: duckypad_list_files: fname=('keymaps', 1) result=[4, 0, 0, 1, 107, 101, 121, 109, 97, 112, 115, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,118 [DEBUG] hid_op: duckypad_list_files: fname=('profile1_windows', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 49, 95, 119, 105, 110, 100, 111, 119, 115, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,142 [DEBUG] hid_op: duckypad_list_files: fname=('profile2_browser', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 50, 95, 98, 114, 111, 119, 115, 101, 114, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,166 [DEBUG] hid_op: duckypad_list_files: fname=('profile3_mypaint', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 51, 95, 109, 121, 112, 97, 105, 110, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,198 [DEBUG] hid_op: duckypad_list_files: fname=('profile4_stream-admin', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 52, 95, 115, 116, 114, 101, 97, 109, 45, 97, 100, 109, 105, 110, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,222 [DEBUG] hid_op: duckypad_list_files: fname=('profile5_WASD', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 53, 95, 87, 65, 83, 68, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,246 [DEBUG] hid_op: duckypad_list_files: fname=('profile6_numpad', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 54, 95, 110, 117, 109, 112, 97, 100, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,270 [DEBUG] hid_op: duckypad_list_files: fname=('profile7_autohotkey', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 55, 95, 97, 117, 116, 111, 104, 111, 116, 107, 101, 121, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,294 [DEBUG] hid_op: duckypad_list_files: fname=('dp_settings.txt', 0) result=[4, 0, 0, 0, 100, 112, 95, 115, 101, 116, 116, 105, 110, 103, 115, 46, 116, 120, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,310 [DEBUG] hid_op: duckypad_list_files: fname=('dp_stats.txt', 0) result=[4, 0, 0, 0, 100, 112, 95, 115, 116, 97, 116, 115, 46, 116, 120, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,334 [DEBUG] hid_op: duckypad_list_files: fname=('profile8_Valheim', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 56, 95, 86, 97, 108, 104, 101, 105, 109, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,406 [DEBUG] hid_op: duckypad_list_files: fname=('config.txt', 0) result=[4, 0, 0, 0, 99, 111, 110, 102, 105, 103, 46, 116, 120, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,430 [DEBUG] hid_op: duckypad_list_files: fname=('key1.txt', 0) result=[4, 0, 0, 0, 107, 101, 121, 49, 46, 116, 120, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:30:55,454 [DEBUG] hid_op: duckypad_list_files: fname=('key2.txt', 0) result=[4, 0, 0, 0, 107, 101, 121, 50, 46, 116, 120, 116, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[... cut ...]
2022-03-09 13:31:46,974 [DEBUG] hid_op: duckypad_list_files: fname=('profile8_Valheim', 1) result=[4, 0, 0, 1, 112, 114, 111, 102, 105, 108, 101, 56, 95, 86, 97, 108, 104, 101, 105, 109, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-09 13:31:47,030 [DEBUG] hid_op: duckypad_list_files: fname=('\x13\x08\x14\x00`/x\x08', 0) result=[4, 0, 0, 0, 19, 8, 20, 0, 96, 47, 120, 8, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

thediveo, 2022-07-16:
Same problem here with the HID error: Read error. On a fresh Kubuntu 22.04 LTS.

tony1661, 2022-08-08:
@thediveo not sure if you are still having this issue but I created an install script on PR #99 

Let me know if that fixes your issue.

thediveo, 2022-08-14:
@tony1661 unfortunately, it doesn't: the same OSError: HID error: Read Error still pops up.

This is on a fresh Ubuntu 22.04.1 LTS installation on x86 64bit. I've cloned the duckyPad repo, then switched into the pc_software subdirectory. Please note that the instructions in app_posix.md are somehow broken, there's no `sudo +x ./duckypad_config_install.sh` doesn't make any sense to me.

tony1661, 2022-08-14:
@thediveo sorry I must have made a type. The command should be `sudo chmod +x ./duckypad_config_install.sh`

thediveo, 2022-08-15:
Why `sudo` a `chmod +x ...`? I can understand `sudo`-executing the script, but not ch'modding it?

tony1661, 2022-08-15:
> Why `sudo` a `chmod +x ...`? I can understand `sudo`-executing the script, but not ch'modding it?

I always sudo chmod incase I don't have permissions to modify the file. 

Not sure why it matters.

Did the script work for you?

thediveo, 2022-08-15:
Yes, the script worked for me, installing everything seemingly correctly. USB devices gets detected. However, the configuration tool can access it and throws the above error message instead.

tony1661, 2022-08-15:
> Yes, the script worked for me, installing everything seemingly correctly. USB devices gets detected. However, the configuration tool can access it and throws the above error message instead.

Strange. Are you on an AMD motherboard?

Also, have you tried a different cable? Is your cable USB A to USB C?

thediveo, 2022-08-15:
> Strange. Are you on an AMD motherboard?

Intel chip set and Intel CPU.

> Also, have you tried a different cable? Is your cable USB A to USB C?

It's the same (braided) USB A-C cable made by some crazy Northern furniture seller that I also use with a recent HP laptop. But ... spoiler alert at the end...

> What is your Output when you type `lsusb` into a terminal?

The relevant output of `lsusb`:
```
Bus 003 Device 010: ID 0483:d11c STMicroelectronics duckyPad(2020)
```

`lsb -tv`:
```
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 9: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0610 Genesys Logic, Inc. Hub
        |__ Port 2: Dev 10, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 0483:d11c STMicroelectronics 
 ```

...and yes, DuckyPad is on a USB3 hub.

> Can you confirm if python3-hid is installed?

```bash
apt list python3-hid --installed
Listing... Done
python3-hid/jammy,now 0.9.0.post3-2build4 amd64 [installed]
```

Now, if I connect DuckyPad to a direct mobo USB3 port, then DuckyPad configurator finally works correctly and uploads the configuration. **It fails as soon as DuckyPad is behind a USB hub.**

thediveo, 2022-08-15:
> Did the script work for you?

When running the script as `sudo`.

tony1661, 2022-08-15:
> Now, if I connect DuckyPad to a direct mobo USB3 port, then DuckyPad configurator finally works correctly and uploads the configuration. **It fails as soon as DuckyPad is behind a USB hub.**

Very interesting. I do know there are a few mentions of hubs causing issues ([example 1](https://github.com/dekuNukem/duckyPad/blob/e5985b002af615fc84d883fe05caff1c1f5da2f4/troubleshooting.md#I-press-a-key-and-nothing-happens), [example 2](https://github.com/dekuNukem/duckyPad/blob/e5985b002af615fc84d883fe05caff1c1f5da2f4/getting_started.md#help-my-duckypad-isnt-working-properly))

I did also see an [issue that is similar to yours](https://github.com/dekuNukem/duckyPad/issues/58) that was fixed with a firmware update. Have you tried the latest firmware?

thediveo, 2022-08-15:
Wasn't sure I had after updating the firmware, so I tried and it unfortunately still don't work behind the USB hub. DuckyPad gets recognized behind the Hub and keypress events sent correctly, but Duckypad configurator doesn't work.

dicekillsyou161, 2022-09-12:
I am also having this same issue with Arch using the hardened kernel (if that is relevant)

tony1661, 2022-09-12:
> I am also having this same issue with Arch using the hardened kernel (if that is relevant)

Have you tried the installer I provided above? I tested it on Manjaro tho, not Arch

enbrown, 2022-10-22:
I'm having a similar problem (on Fedora). It's able to connect and start downloading the current configuration, but bombs on loading one of the key?? files in profile1_windows. If I change `hid_op.py`'s `DUCKYPAD_TO_PC_HID_BUF_SIZE` to be smaller, I can force an error earlier. Rarely, it will occur in the `duckypad_list_files()` function instead.

Somehow a read of `[4, 0, 0, 0, 20, 1, 20, 21, 103, 83, 102, 2, 0, 0, ...]` seems to be popping up in the most recent read prior to the failing one. For example, one time it had an error when loading `profile1_windows/key6.txt`:

1. `duckypad_read_file()` was called to read `profile1_windows/key6.txt`
2. `_read_duckypad()` was called in a tight loop
    1. the first call to `h.read()` seems to always return `[0]`
    2. the second call returns `[4, 0, 0, 0, 20, 1, 20, 21, 103, 83, 102, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]`
    3. this gets converted to `gSf`
    4. since `result[2]` isn't `HID_RESPONSE_EOF`, the loop goes again
    5. again, the first call to `h.read()` returns `[0]`
    6. this time, the second call returns `[4, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]` which has the error code at index 2

This read of `[4, 0, 0, 0, 20, 1, 20, 21, 103, 83, 102, 2, 0, 0, ...]` seems to always precede the read error (even when the error occurs during the `duckypad_list_files()` function.

enbrown, 2022-10-22:
Well, replacing the `_read_duckypad()` code in `hid_op.py` with the version in `duckyPad-profile-autoswitcher` (https://github.com/dekuNukem/duckyPad-profile-autoswitcher/blob/f51eb90c1d6cdf68a329a4c4b12eaf7a0f8a6f2c/hid_example/ex2_read_write.py#L27) seems to make it work without a read error.