dekuNukem/duckyPad #66: Windows 10 + Zen 3 not recognising FW 18.0 or higher

amck1nnon, 2021-08-15:
I have kind of fixed this issue for myself now, but I thought I should probably document that here since there was a fair bit of trial and error - and the solution was different for two machines. Apologies for the stream-of-consciousness style. Between the two machines, there was a decent amount of work to figure out what was going on.

This _might_ be related to the B550/X570 USB issues listed in the release notes... But as far as I can see
- that doesn't seem to have an open issue thread already
- troubleshooting associated with the fix *might have* worked on one of the machines, but it did not contain hardware that is known to be affected by the reported USB chipset bugs
- the issue itself wasn't really consistent with the other AMD chipset/USB issues experienced by myself and others on the internet, wherein there seems to be some kind of sporadic signalling issues that cause devices to occasionally stutter/drop out etc. In my case the device failed to be recognised by Windows in any way - it acted as if it was not plugged in at all

Machine 1: Asus ROG Flow X13 - Ryzen 7 5800H (Zen 3 Mobile)
- Device functioned as expected on firmware 17.0
- No longer lit up or powered on at all following upgrade to 18.0, and Windows did not even seem to acknowledge something had even been plugged in
- Issue not resolved by BIOS update (to be expected - I don't believe any AGESA updates are being issued to address this for Zen 3 Mobile, as it doesn't seem to be affected. As noted above, the most common theory is that it's PCIe 4.0 related - which the mobile variant of Zen 3 doesn't support)
- Appeared resolved following AMD chipset driver update, device then recognised and functioned as expected - but in hindsight (troubleshooting on each device was easily at least 2-3 weeks apart) and considering how it was fixed for my other machine - it may have actually been the change from a USB A port to a USB C port, which presumably the laptop may treat as a different hub?

Machine 2: Asus ROG Strix X570 - Ryzen 7 5800H (Zen 3 Mobile)
- Device functioned as expected on firmware 17.0
- No longer lit up or powered on at all following upgrade to 18.0, and Windows did not even seem to acknowledge something had even been plugged in
- Issue not resolved by BIOS update (latest beta revision for my motherboard contains AGESA 12.0.3B)
- Issue not resolved by latest AMD chipset drivers
- Issue not resolved by the first of two USB hubs I tried (in this case - the hub built into my monitor. Standard USB 2.0 hub)
- Issue not resolved by disabling all PCIe 4.0 support, disabling global C States, disabling XHCI handoff
- NOTE: is was about here amongst the many reboots I noticed that it still worked perfectly in the BIOS, and would turn off partway through the Windows loading screen
- Issue not resolved by disabling driver signature enforcement, safe mode or a Windows repair
- Issue not resolved by trying other USB ports (front panel, CPU controlled, chipset controlled etc)
- Issue finally resolved by plugging into a more advanced USB 3.0 hub with HDMI/Ethernet/other junk; at which point Windows notified me that it was installing drivers for 'duckyPad(2020)' and then functioned as expected
- Following the detection by Windows and the installation/updating of drivers, the device is now detected and functional without the hub, on all USB ports, regardless of AGESA version and the other settings (have since rolled back to last non-beta version)

I don't know enough about USB to say for sure, but having now flicked through everything; it seems like it might come down to some kind of Zen 3 + Windows 10 specific USB driver/configuration issue?

LeoBoost, 2021-09-12:
Working fine here with Firmware 0.19.1, running on a Ryzen 7 5800X on an ASUS Prime X470-Pro - I don't know what kind of conclusions one could draw from that, but I also never encountered the other USB related bugs with this setup.

vdcorduroy, 2022-01-12:
I had same the issue as amck1nnon with Ryzen 3700x & MSI x570 MPG Edge, running Windows 10 on NVME drive with PCIE 4.0.
I tried all of the mentioned as OP, the only thing that finally helped was trick with USB hub. Connecting DuckyPad through hub triggered installation of drivers and it was recognized almost immediately.

closed 2023-05-08 by dekuNukem