dekuNukem/duckyPad #68: Unicode Support

kj4ezj, 2021-09-12:
## What
With duckyPad firmware `v0.19.1` and duckyPad Configurator `v0.13.3`, [Unicode](https://en.wikipedia.org/wiki/Unicode) is not supported.

Specifically, I am unable to:
1. Set a profile name with Unicode characters
   - Example: [`pīnyīn`](https://en.wikipedia.org/wiki/Pinyin)
1. Set a key name with Unicode characters
   - Example: `ˉ`, a [macron](https://en.wikipedia.org/wiki/Macron_(diacritic))
1. Send Unicode characters using the `STRING` operator in duckyScript
   - Example:
     ```duckyscript
     STRING ā
     ```

From the host computer's perspective, non-ASCIII<sup>[1]</sup> characters are stripped from the string. In cases where that results in an empty string, the operation fails or is a no-op. If you inspect the contents of the SD card then you'll find the Unicode characters from the macros are in the `*.txt` config files, but they are not sent to the host.

Worse more, there is no mention of Unicode (or lack thereof) in the [README](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/README.md), [Kickstarter](https://www.kickstarter.com/projects/dekunukem/duckypad-do-it-all-mechanical-macropad), [Tindie](https://www.tindie.com/products/dekuNukem/duckypad-do-it-all-mechanical-macropad), [Writing duckyScript](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/duckyscript_info.md), or [Common Issues / Troubleshooting](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/troubleshooting.md) pages.

## Why
[All languages](https://stackoverflow.com/questions/702899/need-a-list-of-languages-that-are-supported-completely-by-ascii-encoding) besides _maybe_ English and traditional Latin require non-ASCII characters. Even English speakers rely on Unicode to type emoji, arrows, mathematical symbols, and other special characters. For the duckyPad to be inclusive and useful, Unicode support is crucial.

## How
There would be three parts to this work:
1. Supporting Unicode in duckyScript
1. Supporting Unicode on the OLED
1. Supporting Unicode in the "configurator" app

Adding support for the [UTF-8](https://en.wikipedia.org/wiki/UTF-8) Unicode would probably be easiest and make the most sense due to backwards-compatibility and filesystems support. Using UTF-8 string libraries adding maps to `font.c`, and UTF-8 strings in duckyScript would get us much of the way there. I wondered if it is even possible for a keyboard to send Unicode characters to a host, but [QMK seems to have figured it out](https://beta.docs.qmk.fm/using-qmk/software-features/feature_unicode), so the answer must be "yes."

### Memory
This might be more work than I thought when I started writing this issue. -_-

The biggest obstacle is the memory of the STM32F072C8T6 micro [we're using](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/build_it_yourself.md#technical-notes). [Digikey](https://www.digikey.com/en/products/detail/stmicroelectronics/STM32F072C8T6/5051281) and [Mouser](https://www.mouser.com/ProductDetail/STMicroelectronics/STM32F072C8T6?qs=4b8myOmUP%252BsnybMOCvNZeA%3D%3D) spec the SoC out at 16 kB RAM and 64 kB ROM, which is not enough for full Unicode support. The font map alone would be 2<sup>21</sup> * 6 * 10 = 120 MB if all codepoints were supported and used. We could cut it down to 4 MB or so to support languages using the first 16 bits, plus all emoji.

We could:
1. Page in Unicode fonts from the SD card
1. ~~Try to compress the font map~~
1. Use the configuration tool to load only the Unicode characters in-use by the OLED for the current set of profiles onto the micro, since most characters won't be used by any given customer profile set, but this will make manual configuration harder
1. Bump the SoC to something that has enough RAM for future orders, but this won't help existing customers

### OLED
I spent a brief amount of time making sure the OLED supports Unicode. I can't read German, but an OLED equivalent to the one [we use](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/build_it_yourself.md#oled) for sale on [Adafruit](https://www.adafruit.com/product/5228) supports writing arbitrary characters or pixels. In the code, it looks like [`print_keymap()`](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/my_tasks.c#L322) is calling [`ssd1306_WriteString()`](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/ssd1306.c#L236), which calls [`ssd1306_WriteChar()`](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/ssd1306.c#L194), which is capable of [writing arbitrary pixel arrays](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/ssd1306.c#L207) given as fonts. We're passing [`Font_6x10`](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/my_tasks.c#L352) from [`font.c`](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/fonts.c), so adding new characters to this map should be sufficient to support UTF-8.

## Notes
1. Technically the subset of ASCII characters listed in [fonts.c](https://github.com/dekuNukem/duckyPad/blob/ea8681f502395711e48f4435a333ae9958318c29/firmware/code/Src/fonts.c).
1. **Why can't you use duckyScript to enter Unicode characters as shortcuts?**
   I can, but this isn't portable. Furthermore, not all hosts support this for any given character I might want to type. For example, macOS does not seem to support writing all characters needed for [pīnyīn](https://en.wikipedia.org/wiki/Pinyin) without changing keyboard layouts.
1. **Why don't you compile the project with a font map including all the characters _you_ are trying to use?**
   I probably will, I haven't gotten that far. But that isn't a realistic solution for most people.
1. I am very exited about my duckyPad as-is, grateful for an opportunity to contribute by opening an issue, and hopeful that I can contribute code soon.

## See Also
- [Issue 12](https://github.com/dekuNukem/duckyPad/issues/12) - Internationalization
- [The Absolute Minimum Every Software Developer Absolutely, Positively Must Know About Unicode and Character Sets (No Excuses!)](https://www.joelonsoftware.com/2003/10/08/the-absolute-minimum-every-software-developer-absolutely-positively-must-know-about-unicode-and-character-sets-no-excuses/)

I look forward to hearing feedback on this.

dekuNukem, 2021-09-13:
Hi there!

Thank you very much for the feedback! I do agree that unicode support is very important to the internationalization of duckyPad. However, as you pointed out, it would be quite a lot of work to get it implemented properly. And I have a few observations of my own:

* On the current firmware, the RAM is 98% full, and ROM is 85% full, so there probably won't be too much new features to squeeze in without extensive rewrite and optimization.

* duckyPad OLED support the first 128 ASCII characters, and duckyScript itself supports the full 256 extended ASCII characters. The vast majority of duckyPad users are in North America and Europe, where this coverage is sufficient.

* duckyPad emulates a regular USB keyboard and send scan codes to emulate key presses, and I don't think there are unicode keys on keyboards. You can type them out with ALT codes and other shortcuts, but you can do it with duckyScript too.

* I took a look at the QMK support, and it’s a bit clunky to be honest. You’ll need to change the configuration file, edit the source code, re-compile, and still have to do system-level changes to get it to work. Different systems handle unicode inputs differently, so that's another thing to worry about.

In the end, I feel that unicode is a rabbit hole that I'm a bit reluctant to go down at the moment. Although it might be a good idea to add an option in the configurator to convert unicode characters to ALT codes should the user want. It is definitely on the table should I make an updated version of duckyPad with a more powerful processor though.

uckfayer, 2022-03-12:
Due to the USB HID Spec sending actual Unicode characters is impossible and the only workaround is using an OS specific Input Method like Alt Codes or the Compose Key.
One helpful workaround could be a script, website, or modification to the Configuratior to automatically convert Unicode characters to Duckyscript for the the equivalent Alt Code.

closed 2026-02-27 by dekuNukem