dekuNukem/duckyPad #69: Feature request: random input

tnorthcutt, 2021-09-23:
First off: I _love_ my duckyPad! It's fantastic and saves me tons of time.

One of my common use cases is for manual UI testing while I'm doing software development. In these situations, it would be very useful to have some way of randomizing the typing – for instance, I often need to fill out the same form over and over, but with a different name, email address, or other details each time.

In one software application I've used in the past that provides text expansion triggered by a shortcut, I've accomplished this by having it type the current UNIX timestamp, for instance, as a unique string.

Is it possible to expand duckyScript to allow for some sort of random character output, or executing user-defined functions, or something else that would solve this use case?

dekuNukem, 2021-10-03:
Hi there! Random input is a bit of a niche feature, so it's not very high on my to-do list right now. However, you can easily use a input scripting tool like autohotkey or pyautogui to achive what you want, and use a hotkey on duckyPad to trigger it.

tnorthcutt, 2021-10-04:
@dekuNukem That certainly makes sense, thank you for the suggestion! Leaving this open in case you want it as a placeholder for the feature request but I understand if you choose to close it.

closed 2023-02-07 by tnorthcutt