dekuNukem/duckyPad #71: Is it possible to press alpha keys simultaneously?

Nascentes, 2021-09-26:
Trying to create a macro that combines various alpha keys but cannot get them to register as simultaneous presses without millisecond delays.

I've tried:

```
DEFAULTDELAY 1
STRING a b
```

```
DEFAULTDELAY 1
KEYDOWN a b
KEYUP a b
```

Not really sure how else to try it.

dekuNukem, 2021-09-26:
First of all, `DEFAULTDELAY` is the number of milliseconds to wait between **each line** of code. 1ms is way too short for the computer to catch up, the default is around 18ms, and it's better to stick to that for now.

I think `KEYDOWN` and `KEYUP` only work with one key, so try this

```
KEYDOWN a
KEYDOWN b
KEYUP b
KEYUP a
```

Let me know how it goes!

Nascentes, 2021-09-26:
Wow that was quick! No joy on that. Heck, even if I test using modifiers and do...

`CTRL RCTRL` for inputs, it still registers CTRL prior to RCTRL and not simultaneously

dekuNukem, 2021-09-26:
I tried out `CTRL RCTRL` on mine, and it seems to work fine, here is the key history in autohotkey:

```
A2  01D         d    2.61    LControl
A3  11D         d    0.01    RControl
A3  11D         u    0.02    RControl
A2  01D         u    0.03    LControl
```
I think on a keyboard it's not possible to press two keys "simultaneously", there's always a first and second, just with a tiny delay in between. The computer acts when the second key is pressed while the first is still held down. 

Anyway, if you want, feel free to tell me more about what you want to achieve, and maybe I can see what is the best approach.

Nascentes, 2021-09-26:
That makes sense to me. I'll see about pursuing some other options. Mostly trying to set up an accessibility profile for a friend to be able to play some fighter games on PC. May need to pursue using a virtual controller and/or AHK

Nascentes, 2021-09-26:
Can confirm I was able to accomplish this via AHK. Just in case anyone else Googles upon this, here is what ended up working for AHK & DuckyPad:

```
#NoEnv 
; #Warn
SendMode Input
^1::
  {
	Send, {a down}{b down}
	Sleep, 20
	Send, {a up}{b up}
  }
return
```

With my Duckypad key being simply `CTRL 1`

closed 2021-09-26 by Nascentes