dekuNukem/duckyPad #73: Feature request: recognize key_down/key_up on ducky keys

schaeferjoh, 2021-10-04:
Hello and thank you for giving us the duckyPad ! I love it!

I would be interested in recognizing the key_down/key_up events of the duckyPad keys itself.

Something like this: 
  1) Press ducky key : do some stuff
  2) Release ducky key: do something else

I got the idea, thinking about a push-to-talk feature for MS Teams. Your mic would be muted, you press and hold the duckyPad to unmute and speak, and release the duckyPad to be muted again.

I guess duckyScript would need basic if/else functionality for this?

dekuNukem, 2021-10-04:
Hi there! You can use `HOLD` command to make a key behave like a regular keyboard key. It will hold it when you hold it, and release when you release. `HOLD` command can be used with up to two keys, and should be only by itself.

here are more information: https://github.com/dekuNukem/duckyPad/blob/master/duckyscript_info.md#hold-experimental

schaeferjoh, 2021-10-04:
Thanks for your fast answer, but unfortunately `HOLD` will not work for me, at least it doesn't work with MS Teams. The Teams shortcut for mic mute is implemented as a toggle, so holding the keys will toggle repeatedly. For my purpose I would need something like this

`if (key_down) { 
  CRTL SHIFT M // first event
} else if (key_up) { 
  CRTL SHIFT M // second event}`
  
  which would send exactly two events.

dekuNukem, 2021-10-05:
In this case, it might be easier to offload the task to autohotkey. You can put something like `HOLD F20` in duckyScript, which will press F20 when you press the key, and release F20 when you release it. Then you can write a simple autohotkey script to capture F20 and do what you want.

Here are some resources:

https://github.com/dekuNukem/duckyPad/blob/master/getting_started.md#advanced-scripting-with-autohotkey

https://www.reddit.com/r/AutoHotkey/comments/9m4d6k/make_key_activate_on_release_and_press_only/

closed 2022-11-23 by dekuNukem