dekuNukem/duckyPad #75: Receive events on the pad

PaulBGD, 2021-11-02:
Something cool could be to receive events on the duckypad. Like if you receive a notification, you could change the light of a key, or maybe even add something to the screen. I think there's a lot of possibilities with this, although the configuration seems like it could get messy.

dekuNukem, 2021-11-02:
you already can! well sort of...

duckyPad supports receiving HID commands from PC to control it, right now it is used for autoswitcher functionalities such as read/write files and change profiles, I do have plans to add more I'm really busy right now. Still it's definitely not off the table!

https://github.com/dekuNukem/duckyPad-profile-autoswitcher/blob/master/HID_details.md

PaulBGD, 2021-11-02:
Oh interesting! Yeah I noticed in the unicode issue you mention the flash and memory are pretty full right now, I imagine that might have to be improved before more features are added.