dekuNukem/duckyPad #77: Protecting sensitive data entry

jose1711, 2021-11-08:
> duckyPad is NOT intended to be a security device, so use at your own risk.

Despite the warning I am pretty sure many users will be tempted to use DP for automating tasks involving typing passwords, passphrases and what not. Currently this can only be achieved via a password stored in plain text on SD card which is almost on par with a sticky note on a computer screen.

To improve this situation DP could add a new command `SECRETSTRING` which would be a combination of DP serial number (some, password typed during boot (disabled by default) and actual password. As passwords can (and should) contain different sets of characters it would be quite challenging to guess the correct password. That means that the DP password would not be validated at all and it will be only used in macros using suggested `SECRETSTRING` command. Password validation will be carried out by the actual server/service which should mitigate brute-force attacks.

excenter, 2022-05-13:
I'd adore it in the same vein if it had either a fido key or the ability to pass one through.

dekuNukem, 2026-02-27:
I'd rather not go down this rabbit hole for now 😅