dekuNukem/duckyPad #82: Feature request: Reset loop by holding a key

jose1711, 2021-12-07:
`LOOP` command is quite useful but sometimes I am missing a way to reset the loop. I think it would be great if there was an option to reset the counter (and start from `LOOP0`) after holding the same key. As there might be use-cases where key auto-repeat is actually useful perhaps a new command `LOOPH` (loop with reset by holding) could be introduced.

dekuNukem, 2021-12-07:
Actually there is a top secret undocumented command that does exactly that 😅

`LCR` command will reset the loop counter to 0. It was there for debugging purposes and I never removed it.

jose1711, 2021-12-08:
Heh, interesting. Syntax checker will complain but the command indeed works. Also it's useful to know that it resets *all* counters (in all `LOOPx` commands). That means that it is useful only if a user has one key utilizing loops and at the same time can afford to sacrify another key for resets.

closed 2023-02-06 by dekuNukem

asd-ez, 2023-12-28:
@jose1711 Could you tell me about how do you use this `LCR` command? I try to write like this:
```
LOOP0:
VAR $toggle = 1
WHILE $toggle == 1
	TAB 
	DELAY 100
	SHIFT 5
	DELAY 100
END_WHILE

LOOP1:
LCR 
```
But throw error `Unknown command: LCR` and I can't save this script.