dekuNukem/duckyPad #86: Problem with a key color

mobily, 2022-01-04:
Hello @dekuNukem! I found a pretty strange issue, one key has a different color, it's green, but in the duckyPad configurator, it's set to red. Do you know what may cause this problem?

[image]
[image]

dekuNukem, 2022-01-04:
That's very strange indeed!  I wonder if the red element of that RGB LED is damaged, maybe try a few more colours and see if there is any red element on that LED at all. Maybe try a different SD card too to rule out the possibility of file corruption too.

If it is indeed a hardware issue, I can arrange a replacement.

mobily, 2022-01-04:
@dekuNukem it started working correctly again, apparently, an SD card reformat has helped 🚀 thank you, keep it up, I love duckyPad! ❤️

mobily, 2022-01-04:
@dekuNukem it seems to be a problem with the red element of that RGB LED, and yes, it might be damaged 😢

- works when the color is set to cyan:

[image]

- works when it's set to blue:

[image]

- works when the color is to green:

[image]

- doesn't work when the color is set to white, brown, purple, or red:

[image]
[image]
[image]

dekuNukem, 2022-01-04:
yeah that does look like the case, let me know your order number and I can send you another one for free.

mobily, 2022-01-04:
thank you so much, I really appreciate that! ❤️ here's the order number: 300063

dekuNukem, 2022-01-04:
Sure! I'll send it out tomorrow :) Same address?

mobily, 2022-01-04:
yup, same address 😊 thank you once again!

closed 2022-01-04 by mobily

dekuNukem, 2022-01-06:
I have dropped it off, and it should arrive in the usual time :)

mobily, 2022-01-19:
hey @dekuNukem, I got the new board you sent me, thanks for that! ❤️ however, seems it might be broken as well, after connecting to my mac, the SD card isn’t visible, as well as `dfu-util` doesn’t see it, keys don’t react on pressing, and the RGB backlight doesn’t work too

I have tried to use a different either USB-C cable or USB port, and literally, nothing works 😢 although, it boots up! 

[image]

dekuNukem, 2022-01-19:
Hi there! Sorry to hear about the issue, it is rather unusual to have 2 of them be faulty in a row. I would suggest try holding `+` button on the duckyPad, it should go into quickswitch mode, and LEDs should be all white, see if that works. You can also try [re-formatting the SD card and copy back the sample profiles](https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card), or even try one of your own SD cards. It would also be helpful to try it on a different computer, a PC perhaps, or even a phone or tablet.  Let me know how it goes!

mobily, 2022-01-19:
> it is rather unusual to have 2 of them be faulty in a row.

that's what I initially thought as well, it's simply impossible to have that bad luck 😆 

the old board → seems everything works fine, I can connect with the board via the configurator app, `dfu-util` recognizes the device, the RGB backlight works as well (excluding that one key backlight) and when I press a key, a specific action is performed on my mac

the new board → I can't connect with the board via the configurator app, `dfu-util` doesn't recognize the device, the RGB backlight is off all the time, nothing happens when I press a key, however, I can switch profiles with `+` and `-` buttons and go into the quickswitch mode

both boards are using FW 0.19.8, is there any other way to flash FW if `dfu-util` doesn't recognize the device? 

(here's the photo with both boards, the old board is on the left side, the new one is on the right)

[image]

mobily, 2022-01-19:
the new board isn't listed in the connected HID devices
[image]
and the old one:
[image]

dekuNukem, 2022-01-20:
It seems that the new duckyPad isn't getting data through, that's why it is not recognised by the PC. Perhamp you can try cleaning the pins on the USB connector as well as the pins on the microcontroller chip using a dry clean toothbrush or somthing, make sure nothing is stuck between the pins. Maybe try a different able too, but I'm sure you already did that.

Anyway, I can keep sending you more boards until it works 😅 but let's make sure first