dekuNukem/duckyPad #87: HOLD works only one time correctly

benjaminaigner, 2022-01-15:
Dear @dekuNukem ,

I got a duckyPad as a present from my colleague and now I'm adapting it to my personal needs.

Following behaviour I noticed with the HOLD functionality:

**Expected**
I want to switch between windows via `ALT+TAB` (opening the overview), step through the windows
with `LEFT` / `RIGHT`. To keep the selection opened, `ALT` should be hold with the one button, switching
the windows is done with two other buttons.
Following script is used for the `ALT+TAB`:
```
HOLD ALT
TAB
```

**Current behaviour**
The first time it is working, `ALT` is pressed, `TAB` is pressed & released, `ALT` is released on button release.
But the second time, the `ALT` key is still hold, but no `TAB` is sent.

THX for this nice piece of hardware!

dekuNukem, 2022-01-15:
Hi there! Please use `KEYUP` and `KEYDOWN` for this use case. You can [read more about it](https://github.com/dekuNukem/duckyPad/blob/master/duckyscript_info.md#KEYDOWN--KEYUP) here.

benjaminaigner, 2022-01-16:
@dekuNukem 

THX for your suggestion, but I'm not sure if this works as expected:

```
KEYDOWN ALT
TAB
KEYUP ALT
```
Would be my suggestion to use KEYUP/DOWN. Unfortunately, this doesn't work.
To open the window overview, the firmware should press the ALT key as long as I hold the button.

Each time I switch the profile, it works with HOLD as expected, but only one time. Then a profile switch is needed to have it working again.

dekuNukem, 2022-01-16:
`HOLD` command makes a key behave like a regular keyboard key, aka it stay pressed when you press it, and releases only when you release. It can be used for push-to-talk, or WASD gamepad and the like.

`HOLD` command is supposed to be only used by itself, and only with one or two keys. Mixing this command with others will cause unintended behaviour as you mentioned.

What happens is that `HOLD` command caches the key into memory so it won’t have to read the SD card every time you press the key, reducing the latency. So it goes through the script the first time, then it executes from cache only from then on.

For what you want to do, you can probably try something like this:

```
KEYDOWN ALT
KEYDOWN TAB
LEFT
LEFT
KEYUP TAB
KEYUP ALT
```

Might need some delay inbetween, but that should be the gist of it, feel free to try it out and let me know how it goes!

benjaminaigner, 2022-01-17:
THX for the clarification!

Maybe another way to implement this key combination:
Would it be possible to introduce a `WAIT_RELEASE` command which stops the button script execution until the button is released again?
(just a suggestion)

In this case, a script could look like:
```
KEYDOWN ALT
TAB
WAIT_RELEASE
KEYUP ALT
```
Anyway, I will try different ways to solve my problem.

closed 2022-11-23 by dekuNukem

benjaminaigner, 2022-11-23:
@dekuNukem THX, I'll update my pad!