dekuNukem/duckyPad #88: No timeout set on firmware/release check

Nick-Watton, 2022-04-19:
Hello!

OS: Windows 10
Using included USB-C cable

Today there was an issue with GitHub codespaces, which impacted the duckyPad configurator application.
Upon clicking Connect, the application would hang for 2 minutes while it checked for new releases and firmware, I assume a default of 1 minute per call.
Once the timeouts had finished, I was able to configure as normal. However saving changes proved a bit problematic - it did update the pad but I also encountered this error during saving.
[image]

I see you implemented a timeout in `is_internet_available()`  in `check_update.py` but not for the GitHub checks. 
I would say the same probably needs to be implemented for these other checks so that the application can function normally without GitHub if necessary.

dekuNukem, 2022-04-19:
good catch! I didn't really expect github would go down like this. I'll add it.

As for the error, looks like it didn't find duckyPad. Was it able to connect and load profiles in the first place?

Nick-Watton, 2022-04-19:
After the timeouts it loaded the profiles correctly, yes.
Saving did not immediately update the board. It went though "Saving.." to "Saved!" in the text next to the connect button, but the board did not update for ~55 seconds after initially clicking save (and only sometimes throws the save-error). Only then did the text update to "done". I'm not sure if there are more calls to GitHub during the autobackup (I can see one [here](https://github.com/dekuNukem/duckyPad/blob/master/pc_software/duckypad_config.py#L1261) but I haven't thoroughly checked the impl), but I think it's probably related to the GitHub issues and would likely be resolved by implementing a timeout on those calls.
Thank you for such a prompt response!

closed 2022-11-23 by dekuNukem