dekuNukem/duckyPad #92: DIY not connecting to pc

VelocityK9, 2022-06-17:
I decided to go to DIY route and made one myself. After soldering everything and plugging into the pc to flash it nothing happens. Holding the dfu button on powerup by my testing just sets the TX and clk to 3.3V

dekuNukem, 2022-06-23:
Hi there! Make sure the chip is in the correct orientation, and all the pins are connected with no solder bridges between them.

If you have a ST-Link programmer you can try hooking it up to the header on top-right of the PCB and see if it connects. Steps are below:

https://github.com/dekuNukem/STM32_tutorials/tree/master/lesson0_intro_blinkLED#hookup

Also, if you got the STM32F072C8T6 chip from ebay or aliexpress or similar, it might have been re-badged from a similar but different part. I once had a STM32F072C8T6 that is actually STM32F071C8T6, which is almost the same but does not have USB functionality, and therefore does not work with USB DFU.

VelocityK9, 2022-06-26:
I'll order a ST-Link and see if I can do anything with it. Since it was ordered from Aliexpress theres a high chance its a "fake". If so do you know where I can buy them in eu or salvage them from boards similar to [this](https://www.aliexpress.com/item/1005003880238896.html?).

dekuNukem, 2022-11-23:
Cleaning up the issues, let me know if you have any further questions!

closed 2022-11-23 by dekuNukem