dekuNukem/duckyPad #93: DuckyPad OLED screen freezed when starting up PC

SnailMango, 2022-06-18:
Hello! My DuckyPad started having this issue since a couple of months.
When I startup my PC, it lights up normally, but when I try to switch profile using the - and + buttons, the screen is freezed on the first functions list. In order to make it work again, I have to disconnect and reconnect the USB cable. If I push the reset button on the side, the screen turns off and it won't work anymore if I don't disconnect and re-connect the cable.
I already tried re-flashing latest firmware and changing cable but the problem is still the same.
What can I do? I'm attaching a video of the problem.

https://user-images.githubusercontent.com/64862343/174437614-ad5e09dc-cf4d-406d-bfab-43790c2fdb29.MOV

dekuNukem, 2022-06-18:
Hi there! Looks like the OLED screen is acting up a bit, re-seating the ribbon cable usually fixes it, you can find the instruction here:

https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#the-oled-screen-isnt-working-properly

let me know how it goes!

SnailMango, 2022-10-26:
Sorry for the super old reply but I just logged back into Github and realized I forgot to reply! However it really looks that reseating the ribbon cable fiexd the screen text problem! Thanks!

closed 2022-10-26 by SnailMango