dekuNukem/duckyPad #94: Closing duckypad config while saving causes partial shortcut loss

fechu, 2022-07-14:
Hi all, 

I just had the case where I was saving scripts to the duckypad via duckypad config. I thought I had moved my focus to the browser where I wanted to close the current tab with CMD+W. That closed duckypad config. After opening again and loading the config part of the scripts are missing now. 

Suggested solution: 
Show confirmation dialog when closing duckypad config if a saving operation is running. 

@dekuNukem If you think this is a good improvement I'm happy to dig into the code and implement it. What do you think?

null0byte, 2022-07-15:
+1 up vote for this fix. I closed the Config app forgetting it was saving and lost some minor changes. No massive issue but good UI fix if implemented.

dekuNukem, 2022-11-23:
added in the latest update

closed 2022-11-23 by dekuNukem