dekuNukem/duckyPad #104: Middle key not working after profile switch

EthanDarling8, 2022-10-05:
- Issue: 
    - After switching between my two profiles the middle key quits working completely.
    - No other key has this issue, only the middle key (3rd row from the top and 2nd column from the left).
    - Currently the hotkey is ALT KP_7. Usually when using a numpad hotkey with duckypad it will type a symbol into any text input. I noticed that when the issue happens no symbol is typed; other keys that use numpad hotkeys will type a symbol.
- What I have tried: 
    - Changing the hotkey that that middle key runs.
    - Checking that the key is seated correctly and the pins are not bent.
    - Switching profiles manually and switching profiles using auto switcher. Both have the same results.
    - Adding delay in various ways.
    - Updating to the latest firmware, updating the duckypad config application, and updating the duckypad auto profile switcher.
- What seems to fix it:
    - Not using the middle key. This is obviously not a good fix.
    - After switching profiles and seeing that the middle key is not working if the reset button is pressed (the 2nd button down on the right side of the duckypad) the middle key starts working again. This is not a good fix because the reset button has to be pressed every time a profile is switched.

Any help is appreciated, let me know if you need more information.

dekuNukem, 2022-10-05:
Hi there! That's a rather strange behaviour, since the middle key works on the first try, I think there’s nothing wrong with the hardware. 

I would suggest reformat the SD card and copy the files back. If you have a spare SD card try using it just to make sure. 

[See this guide](https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card) for steps. "I want to use my own SD card!" Section.

Let me know how it goes!

spants, 2022-10-14:
i had a middle key problem but noticed that one contact leg on the key was shorter than the other. I must have broken when installed. I soldered a short wire on that leg and it works again!

EthanDarling8, 2022-10-14:
Thanks for the help. I tried reformatting the SD card and also tried using a different SD card. Still getting the same behaviour. It's definitely not hardware related, even put in a different switch to test that. Maybe getting rid of the profile and making a new one will fix it. I will try that.

dekuNukem, 2022-10-14:
Actually, you can also try going back to an eariler firmware, maybe it was a bug in the new firmware.

The instructions are below, scroll down and use the 0.19.8 version.

https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md

EthanDarling8, 2022-10-14:
> Actually, you can also try going back to an eariler firmware, maybe it was a bug in the new firmware.
> 
> The instructions are below, scroll down and use the 0.19.8 version.
> 
> https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md

I'll give that a shot as well.

EthanDarling8, 2022-10-21:
Downgrading to 0.19.8 fixed the issue. I'll try upgrading to 0.20.1 just to verify that it is the new firmware but it seems to be fixed now.

dekuNukem, 2022-10-28:
I was able to reproduce the bug, and it was (hopefully) fixed in the latest firmware.

EthanDarling8, 2022-10-30:
> I was able to reproduce the bug, and it was (hopefully) fixed in the latest firmware (0.20.2).
> 
> Try it out here: https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md

Awesome, thanks! I'll give it a go and let you know if it works.

EthanDarling8, 2022-10-31:
Looks like the 0.20.2 release fixed it. Thanks!

closed 2022-10-31 by EthanDarling8