dekuNukem/duckyPad #106: layout issue

smarTHh2019, 2022-10-11:
Hi,
duckyPad has a **layout error** e.g. "y" is a "z" (a common EN-DE keychange). I added the dpkm_German.txt. No change. In the configurator under "Keyboard Layouts...", I cannot remove "English (US)" for all other languages adding and removing works. Adding German (via the configurator option or .txt file) works but doesn't change anything and after "save" and "connect" the German setting is even gone. The SD card has the file dpkm_German.txt placed in keymaps/.
Running the script in the configurator works z=z but running it via the keys fail z=y.
Versions: 0.13.3 and 0.20.1 used with Win10 and Win11.
Best, Thomas

dekuNukem, 2022-10-11:
Hi there! You need to select the keymap after adding it.

Unplug duckypad, hold down TOP LEFT key while plugging it in, and select the keymap.

smarTHh2019, 2022-10-11:
Hi dekuNukem,
yes, that fixed it. Many thanks

closed 2022-10-11 by smarTHh2019