dekuNukem/duckyPad #109: STRINGLN not working after upgrade to firmware version 0.20.2

seawarrior181, 2022-11-20:
Hi,

I've learned today that the new STRINGLN command isn't working after upgrading the firmware to 0.20.2.  I make extensive use of the construct:
STRING macro text goes here
ENTER
...so I am very interested in the STRINGLN functionality.  What do I need to do to activate this command?

A snap of my duckyPad configurator v0.13.3 is included below.

[image]

Notice that Code Check indicates **Error(s) found!**

Best,
Dan

dekuNukem, 2022-11-20:
Just press `Save` anyway and it should work on the pad itself! I just haven't released the software update yet 😅

seawarrior181, 2022-11-20:
ok, that worked, thanks.

closed 2022-11-23 by dekuNukem