dekuNukem/duckyPad #116: Add command to display a bitmap on OLED screen

jose1711, 2023-02-17:
duckyPad's display is both small and monochrome so I think it would make sense to have a command to display an arbitrary bitmap (e. g. a logo or a text in custom font and/or additional graphical elements). The bitmap would be stored on SD card (e. g. in `img/foo.bmp`) and the command syntax could go
```
OLED_BLIT foo
```

jose1711, 2023-03-10:
Experimental implementation is at https://github.com/jose1711/duckyPad/commit/0fd8e1cb9756375cb1fe5f7056e76e6c55ba2684

jupiterbjy, 2023-08-29:
Totally love this idea - could be used as a startup splash screen sort of.

Feels like even a command to set a single pixel instead of character - people could do so many things with it. With all that features in duckyscript3 one could draw animation out of it, who knows!