dekuNukem/duckyPad #122: duckyPad configurator should improve argument checking

jose1711, 2023-03-26:
configurator (DuckyScript3 version) evaluates the following line of code as valid (`Code OK`):
```
DELAY 99999999999999999999999999
```
yet it reports an error on compilation (`int too big to convert`) as it cannot fit 16-bits (unsigned short). It would be better if such an error is detected prior Save operation.

mpieper, 2023-08-24:
In ds_syntax_check.py -> line 65, add a case for the int cast?

You'd have to do due diligence that there are no cases where a uint32 or other type of variable are picked up from the parser.

```
def check_one_arg(pgm_line):
	split = [x for x in pgm_line.split(' ') if len(x) > 0]
	if len(split) != 2:
		return PARSE_ERROR, "only one argument allowed"
	try:
		value = int(split[1])
                if value >65535:
                    return PARSE_ERROR, "uint16 value exceeded.  limit to range 0-65535"  
	except:
		return PARSE_ERROR, 'invalid value'
	if value < 0:
		return PARSE_ERROR, "value can't be negative"
	return PARSE_OK, ""
```

dekuNukem, 2026-02-25:
Done in latest version 😅