dekuNukem/duckyPad #128: Issue Updating to Firmware 1.2.0

nokternlix, 2023-05-12:
I downloaded the latest firmware version 1.2.0 from github and followed the instructions on how to update. I put the duckypad into DFU Mode. I used the command dfu-util --device ,0483:df11 -a 0 -D /path/to/duckypad_firmware.dfu (changing the path of course) and it went through fine. No errors parsing the file and dfu-util said everything completed successfully. I unplugged the USB-C cable and plugged it back in to reboot the device. The keypad lit up and it showed my profiles. i could even cycle through the profiles and the screen changed appropriately along with the colors of the keypad. the problem was, none of the script files that were bound to the keys actually executed. There were no errors, just nothing happened. I verified that dfu-util could see my duckpad. It showed the same usb device id that is listed in the update command.

This happened both times i tried to update the firmware on my duckpad

I'm running Linux Mint 21.1 with all the updates dfu-util version 0.9-1.

dekuNukem, 2023-05-12:
hi there! this major update changed how script is executed, so you need to use the latest configurator to load and then save again. then it should work!

you can find it here: https://github.com/dekuNukem/duckyPad/releases/tag/1.3.0

nokternlix, 2023-05-12:
welp, i feel pretty dumb. i guess i was too impatient to let the configurator detect the duckypad as it didn't appear to be doing anything and i couldn't click on the 'connect' button at the time. i apologize for bothering you with my stupidity. thank you for your  quick response though, i really appreciate your help.

dekuNukem, 2023-05-12:
no problem! glad it's sorted out!

closed 2023-05-12 by dekuNukem

lukasgierth, 2023-07-03:
I was just updating my duckypad and am running into the same problem. Is there a way to resave the script without the configurator app? I'm using Fedora Silverblue and using the configurator is not really an easy option, i'm doing all configuration by hand in the text files. So what exactly needs to be done?
@dekuNukem

dekuNukem, 2023-07-06:
Hi there! Duckyscript 3 introduced a lot of advanced features such as variables and flow control statements, so with the new update, the scripts are now now compiled into a binary file to be executed on a virtual machine on duckyPad itself.

First of all, if you don't need the new features, you can always go back to [0.21.0](https://github.com/dekuNukem/duckyPad/blob/master/firmware_updates_and_version_history.md#firmware-version-history), and everything would be like before.

However, if you do want to use the new features, you'll need to compile the scripts into `.dsb` files and include them along with the original `.txt` files. The [instructions are here](https://github.com/dekuNukem/duckyPad/blob/master/duckyscript3_bytecode_vm.md), and you can take a look at the [sample profiles](https://github.com/dekuNukem/duckyPad/raw/master/sample_profiles.zip) as an example.

That does mean you'll need to execute a python script though. If it's not possible on your distro, then you would need to use a different comptuer.

lukasgierth, 2023-07-07:
Thanks for your answer! I see, ok.
Then i will for now keep using the old version without DS3 support and maybe upgrade later when i want to use the newer functions (and compile the dsb files manually).

Yeah,  it's not that convenient to run the python configurator in Silverblue. Fedora Silverblue has an immutable root filesystem so i either need to overlay the needed extra python modules or install them locally with pip. And tkinter is the culprit here since the version installed with pip doesnt work as intended and i dont want to overlay it only for the configurator to configure my stuff once ;) But the manual dsb file creation works for me just fine.