dekuNukem/duckyPad #140: Converting to wireless

rafcio39, 2023-07-28:
is it possible to convert it to wireless control? adding a battery and the ability to charge it?

mpieper, 2023-08-24:
This details just the basics to get a wireless solution

https://www.youtube.com/watch?v=o-wbPVkkyWg  

To actually fork the project over to wireless would involve a uC port to a different stm-chip w/ bluetooth, some reconsiderations over battery consumption / display selection, adding lipo-charging circuitry and some semblance of short circuit protection / fire mitigation, redoing the casing / enclosure, and then firmware changes for pairing / interfacing with the bluetooth module.  That's a good (squints, licks thumb and holds it at contractor's arms length) 3 months worth of work if you know what you're doing.  A veritable purgatory if you don't.

dekuNukem, 2026-02-25:
The latest version now supports bluetooth!

https://dekunukem.github.io/duckyPad-Pro/README.html