dekuNukem/duckyPad #142: No keys working on my duckypad

IcyPenguinCube, 2023-08-16:
I updated the firmware to version 1.3.0, and the app on my computer to version 1.4.0, and my duckypad no longer functions when i press a key. If i change it to a firmware that is 0.9.0 or older, it works, but a lot of the keys give error messages that did not used to.

dekuNukem, 2023-08-16:
Hi there! Use the latest firmware and configurator, load up the profiles in the configurator, then click `Save` to write it back.

It will write some new files to duckyPad and make it work again!

IcyPenguinCube, 2023-08-16:
I am currently using the newest of both, and clicking save crashes the configurator. Thanks

dekuNukem, 2023-08-16:
does it print out debug messages in the terminal window when it crashes?

IcyPenguinCube, 2023-08-16:
it says "first pass OK!
second pass OK!" but then stops

IcyPenguinCube, 2023-08-16:
[image]
[image]
Here are the images of the configurator and terminal windows.

dekuNukem, 2023-08-16:
hmm that's strange, looks like it completely hangs. it's 5am in UK right now, so I'll be going to bed soon and get back to it later today.

In the meantime, maybe you can try making a backup by copying everything off the SD card, and format it in FAT/FAT32, and start fresh, see if it works that way.

IcyPenguinCube, 2023-08-16:
ok, thanks. I will try that and let you know if it works.

IcyPenguinCube, 2023-08-16:
I backed up all of the files, wiped the drive, added them back with a FAT format, and the keys on my duckypad still do nothing.

IcyPenguinCube, 2023-08-16:
clicking the save button gives the same result in the configurator. I tried using the FAT32 format, and nothing happens. I am still able to cycle through the different profiles, but the animations are gone, and pressing a key does nothing.

dekuNukem, 2023-08-16:
i meant making a backup, format the card, don't copy anything back and start fresh. See if the configurator works this way.

IcyPenguinCube, 2023-08-17:
it works after i did that. I will add all of my profiles back. Thanks!

IcyPenguinCube, 2023-08-17:
I copied and pasted a profile from the old backup, then it stopped working again. I wiped the SD card and formatted it again, but now the configurator crashes whenever i click connect, and leaves nothing in the terminal.

dekuNukem, 2023-08-17:
Are you using AMD motherboard by any chance? they have rather buggy USB chipsets.

If so, try using a USB hub, or try a few different USB ports. Also maybe try a different cable or a different comptuer as well with the empty SD card.

IcyPenguinCube, 2023-08-17:
My laptop has an Intel chip, so i dont think so. I have tried using it on another computer and with a different cable and i get the exact same results.

dekuNukem, 2023-08-17:
in that case the final possiblity is that the SD card might have corrupted. if you have a spare micro SD card, try format it the same way and see if it works.