dekuNukem/duckyPad #145: Add Multiple key support back in EMUK (or reviving the HOLD)

jupiterbjy, 2023-08-29:
## Suggestion

Either of this suggestions would make my duckypad usable again, or I might need to downgrade firmware (hope it works!)

1. Add back `HOLD`'s 2 (or more the better) key support in `EMUK`
    - Might break compatibility?

3. Allow use of multiple `EMUK` lines in single button

5. Revive deleted `HOLD`

---

## Encountering issue

I was delaying update of firmware and just recently updated, and found out `HOLD SHIFT SPACE` is now `EMUK SHIFT SPACE` and syntax is no longer valid.

Previously `HOLD` command could held up to 2 keys:
https://github.com/dekuNukem/duckyPad/blob/a2531bf4b8fe2d8dca877a265897d728ad84f521/duckyscript_info.md?plain=1#L352

Now, since `EMUK` don't support this and even worse there can't be 2 `EMUK` in a button, I have no way of setting `Shift` + `Space` hold down hotkey for arbitrary canvas rotation action, which looks like this:

[image]

One could edit the hotkey in software they use but there's some softwares that does not allow this - and it doesn't sounds right to rely on AutoHotkey to reassign it just to make it reassigned on duckypad.

---

## Attempted workaround

Since now Variables are added, I tried using variable & loop:
```
KEYDOWN SHIFT
KEYDOWN SPACE

WHILE $_READKEY != 0
    DELAY 50
END_WHILE

KEYUP SHIFT
KEYUP SPACE
```

Which actually does not work, as issue indicated on #144 I can't maintain the loop.

jesusfchavarro, 2024-09-13:
a little late but you can assign a button to do:
```
KEYDOWN SHIFT
KEYDOWN SPACE
```
and other to do 
```
KEYUP SHIFT
KEYUP SPACE
```

jupiterbjy, 2024-09-14:
@jesusfchavarro sadly but surely that isn't an option.

Try using shift key like that in everyday use, using one key as press and other key as unpress or making it as toggle button - it will be fairly obvious you can't unconsciously use it. This hotkey, is just as frequent, short bursted thing that should only work during pressed.

dividium, 2024-09-17:
I second this. In my case I would like to use PowerToy's Push-to-talk and it requires (in my config) WINDOWS SHIFT Z.

I've tried this:
```
KEYDOWN WINDOWS
KEYDOWN SHIFT 
KEYDOWN Z
EMUK Z
KEYUP Z
KEYUP SHIFT 
KEYUP WINDOWS
```
But the commands beyond the EMUK do not run. After the first press (that behaves correctly), subsequent presses treat is as just Z.

I also tried using a WHILE loop, but it seems as though the script repeats anyways if the key is held down.

A nice flexible approach would be the ability to switch to running one script on KEYDOWN and another on KEYUP. That method would afford a lot of flexibility and might be easier to implement? Might also need the ability to disable key repeat.