dekuNukem/duckyPad #149: Can't connect to DuckyPad on Windows 10 - duckyPad configurator shows "Not responding" after pressing Connect button

raleighlittles, 2023-12-17:
I just assembled my duckyPad and connected it to my Windows 10 PC via the USB-c cable. The screen turns on and shows the message:

> Please Insert SD Card
> For instructions, see duckyPad.com
> FW V1.3.0

(I found this to be odd since there's already a microSD card in the device, as I installed it when I was assembling it.)

 I then downloaded duckyPad configurator v1.63 (the latest), then clicked "Connect" in the application. Nothing happens after several seconds and the window shows "Not responding". The debug log also doesn't show anything.

[image]

[image]

dekuNukem, 2023-12-17:
Hi there! If the screen shows "please insert SD card", then it means the SD card isn't detected.

I would suggest use the SD card reader and give the card a full format and see if it is recongised. 

You can also use your own SD card if you have a spare.

For detailed steps, please take a look at `I want to use my own SD card!` section of this guide:

https://github.com/dekuNukem/duckyPad/blob/master/troubleshooting.md#i-want-to-use-my-own-sd-card

raleighlittles, 2023-12-17:
@dekuNukem Thank you for the quick response, I'll try formatting the SD card and then re-running the steps.

Also just a heads up, there is a small issue with the Linux installation script, it looks like the file endings were corrupted by Windows:

[image]

This is the error I got when I tried to run the script on Linux.

> zsh: ./linux_install.sh: bad interpreter: /bin/bash^M: no such file or directory

After running `dos2unix` on the file, it was fixed.

raleighlittles, 2023-12-17:
@dekuNukem Formatting the SD card worked, so I'll close the issue now, thank you!

closed 2023-12-17 by raleighlittles